pichenettes/eurorack #7: Yarns upload_combo_jtag hex file with following error

altitude909, 2016-10-08:
[yarns_bootloader_combo.hex] checksum : 0x006052B3 
17:28:32 : Error: File corrupted. Two or more segments defines the same memory zone.
17:33:31 : [yarns_bootloader_combo.hex] opened successfully.
                  Address Ranges [0x08000000 0x080010D8] [0x08001000 0x0800F350]

beeva-luisgonzalez, 2016-10-08:
Weird a friend of mine burned 2 yarns,  a couple weeks ago. 

The tool chain has changed? What method of uploading/compiling are you using? I have no problems with the mutable dev environment project (vagrant)

altitude909, 2016-10-08:
Vagrant environment, repo up to date.   It builds the bootloader, main hex files, and the combo hex but when loading with the STM software it throws up that error.  Looking at the files, the bootloader is does overlap the main hex

pichenettes, 2016-10-25:
Compiler version issue, causing the bootloader size to go over 4096 bytes. For now, you can comment one bit in the bootloader to save a few bytes (such as disabling the "snake" animation, in the if condition on ms_clock & 0x3f).

closed 2016-10-25 by pichenettes