pichenettes/eurorack #9: wrong value on schematic braids_v50

thopman, 2016-11-15:
r39 is incorrectly marked as 120K, thus only giving about 2.07V at the adc input when setting the coarse pitch knob to max. It should be 100K.

pichenettes, 2016-11-15:
Not an issue.

The correct value for R39 is 120K as stated in the schematic and BOM.

If R39 was set to 100k as you suggest, then the ADC input would be 2.5V when COARSE would be turned fully CW; and the module would stop responding to positive voltages applied to the V/O input. Using R39 = 120k gives some "headroom" to apply positive transposition (up to one octave + one fifth) on top of the pitch set by the COARSE knob.

thopman, 2016-11-15:
Ah i see, makes perfect sense. Thought it was an issue, because the pitch
knob is the only knob not reaching full 2.5V. Thanks for the reply.

Thomas Hopman - TomaTek - Paradiso - SureShot Studio
+31 (0)6 17115051

closed 2017-10-10 by pichenettes