pichenettes/eurorack #19: Yarns INIT resets calibration settings

dturner, 2019-03-16:
Summary: On Yarns if INIT is chosen the calibration settings are lost. 

Steps to reproduce: 

- Connect a multimeter to Yarns CV1 output to measure the voltage
- Calibrate Yarns CV1 0V voltage (this doesn't need to be exactly 0V, just some different value than is currently shown on the multimeter). 
- Save calibration settings
- Choose INIT (to get a default program)
- Send a C1 note to Yarns
- Check the value shown on the multimeter

Expected result

- Multimeter shows the calibrated value which we just set

Actual result

- Multimeter shows the original value before calibration

Note that the calibration settings can be saved by saving a program after performing calibration. For example, if you save to P1 then the calibration settings will be restored on boot. It's just using INIT which wipes them. 

It may be that this is as designed, however, I wanted to check as it seems like resetting the calibration just to get a default program (using INIT) is not expected behaviour.

dturner, 2019-03-16:
Full disclosure, this is a DIY build. Please close if you think there's no likelihood of it affecting other devices (and sorry for the noise).

pichenettes, 2019-03-18:
Thanks for the bug report! This is fixed.

closed 2019-03-18 by pichenettes

dturner, 2019-03-18:
Wow. Thank you so much émilie. Tested and fixed :D