pichenettes/eurorack #24: How do you fabricate a custom faceplate?

jakebromberg, 2019-08-23:
I've never built a modular synth before but a faceplate seems like a must have. What's your method of choice?

pichenettes, 2019-08-23:
Your question is not an issue with the codebase. This is not a forum.

closed 2019-08-23 by pichenettes

jakebromberg, 2019-08-24:
Thanks for digging deep into your well of emptiness and baring your lack of empathy. Could you direct me to the forum?

jasonnolan, 2019-08-24:
> Thanks for digging deep into your well of emptiness and baring your lack of empathy. Could you direct me to the forum?

https://forum.mutable-instruments.net

timchurches, 2019-08-24:
> Thanks for digging deep into your well of emptiness and baring your lack of empathy. 

Unbelievable rudeness and entitlement! An apology to pichenettes is necessary.

> Could you direct me to the forum?

With that sort of attitude, you won’t be welcome there.

aroom, 2019-08-24:
Unbelievable indeed 😳

jakebromberg, 2019-08-24:
Thanks, @jasonnolan!

jakebromberg, 2019-08-24:
This whole thread is pretty foreboding. Everyone likes or loves the comments in which I'm brusquely cast aside and labelled "entitled", and down voted the comment where I call that out. 

I'll ask that you folks measure up to your humanity and be more mindful and kind there than you've demonstrated here. May you be well, happy and patient.

rcrogers, 2019-08-24:
Hi Jake, long time no see!  I think this might have gone better if you'd taken a few minutes to research how this community operates (how issues/PRs are used, whether there's a forum, etc), or at least made clear that you'd tried -- in other words, showed regard for the way this community operates, and the time of whoever's fielding your questions.

aroom, 2019-08-24:
I’m vegan by the way.

jakebromberg, 2019-08-25:
Hey @rcrogers! Thanks for chiming in and elevating the discussion. 

I'd like to disagree with you on a couple points. To your comment that I should research how the community operates—I completely agree! In fact, that is 100% why I posted here to begin with. Short of some link to the website or some (misguided) rule that issues should only relate to the codebase, filing a ticket here is the next best thing. It is, in fact, my usual way of getting my bearings in any new situation on GitHub or work and it always works for me (except this one time).

More specifically, though, I think this thread would have gone way better had @pichenettes not failed in his role as owner and steward of this community, and bothered to at least provide a link. He took my earnest interested and slapped it in the face. @pichenettes, you're in charge here, and your actions and words carry more weight than anybody's. Your careless attitude resounds in the self-serving comments that followed. If I were you, I'd write a code of conduct that everyone in this community—yourself and mine included—adheres to. Email me when you'd like to get started.

pichenettes, 2019-08-25:
My take on this: I'm on vacations, only an old smartphone. Receiving an email reporting an issue with the codebase made me anxious as hell because I knew I wouldn't be able to fix the bug. It was a trivial issue, so I gave the shortest answer I could type. On the mutable-instruments site the "community" link redirects to the forum.

Things might have been different in other places on github because they are community projects.  This is not the case here, the is not a collaborative, community project. The community is the forum and the rules are there.

My message was dry, but not demeaning – unlike some of your own words.