pichenettes/eurorack #28: Instructions for uploading firmware using OSX

catkins, 2020-05-28:
Hi there! Love your work.

I was wondering whether you though it could be useful to folks to include instructions for uploading firmware using OSX without needing to use the Vagrant dev environment.

For me, uploading firmware to Peaks and Braids was relatively straightforward using an STLink-V2 and an [Olimex ARM-JTAG-20-10](https://au.mouser.com/ProductDetail/olimex-ltd/arm-jtag-20-10/?qs=DUTFWDROaMbVQp3WoAdijQ%3D%3D&countrycode=AU&currencycode=AUD) adaptor.

Requires [Homebrew](https://brew.sh/)

```sh
# add PX4 formulae to homebrew
brew tap PX4/homebrew-px4

# install toolchain
brew install stlink open-ocd px4/px4/gcc-arm-none-eabi-48

# configure make scripts to point to the correct ARM toolchain
export TOOLCHAIN_PATH="$(brew --prefix px4/px4/gcc-arm-none-eabi-48)"/

# compile bootloader
make -f peaks/bootloader/makefile hex

# upload to device
make -f peaks/makefile upload_combo_jtag
```

lijon, 2020-06-17:
I just tried this here and it works fine!

djdoz, 2021-07-08:
Thanks catkins! This works well for me!