pichenettes/eurorack #33: Ripples V0.2 files availability

searingmoggy, 2020-12-17:
Hello! I found among my stuff a Ripples' PCB V0.2 which I started populating years ago (2013) and then abandoned. Willing to resurrect the project, though, I couldn't find any .brd file and, of course, no BOM. I know that's pretty obvious, considered we're at Ripples' V4.0, but I guess that old version should still work so… could anyone please forward me the matching files, if still available? Also, was there anything to pay attention to while putting this project together (i.e. labelling mistakes on the soldering mask, particular trimmings required…)?

Thanks a ton to whoever will chime in, best wishes to everyone!

pichenettes, 2020-12-17:
Same design as v4.0 but with a different type of jacks! No labelling mistakes.

closed 2020-12-17 by pichenettes

searingmoggy, 2020-12-17:
Thank you so much, Emilie!