pichenettes/eurorack #35: Surge and Clouds?

baconpaul, 2021-01-29:
Hi!

I'm one of the maintainers on surge (https://surge-synthesizer.github.io/)

We were chatting today about expanding our FX and realized that Clouds could fit very nicely into surge. Surge is FOSS GPL3 software so from a licensing perspective on the engine I think we are good, and Andrew's VCV rack module makes the API really clear. So I'm debating giving it a whirl.

But before I did that, wanted to check a couple of things

1. Even though it's all GPL3 and we could do it, doesn't mean you would want us to. Any objections or concerns?
2. I know you are careful about your brand - rightly so - and if I understand, the name mutable is trademarked and not open and so forth. That's super smart and we have been very careful about licensing and IP issues in Surge! If we did include clouds, is there a way we would name the FX or credit you which would be appropriate?

In our 1.8.0 release we imported 50+ Airwindows FX into Surge after chatting with Chris, for instance, and pulled in the filters from OBX-d, and ended up putting them in the in-plugin about page as below.

In any case, thanks for making great open source audio tools and for considering this issue!

Best

[image]

pichenettes, 2021-01-29:
1. No objection or concern.
2. I'd recommend naming your FX "Granularizer", "Texture synthesizer", or
anything generic.
3. The way you have credited the other authors is perfect, you can do the
same for me.

baconpaul, 2021-01-29:
Wonderful, thanks you!

mkruselj, 2021-01-29:
That was a super fast reply! Thank you 🙂 

Emilie, personally I love the idea of calling the effect Nimbus, since it's Latin for cloud. Is that "too close to home", or would you be alright with thaT?

pichenettes, 2021-01-29:
That name is okay!

>

jamesbretz, 2021-01-29:
I had not realized Surge went open source, and this conversation has just made my day. 😃

baconpaul, 2021-01-31:
Just FYI, we pushed this to our nightlies last night and surge users are starting to test and enjoy it already. Thank you for providing great (and easy to integrate) free and open source software.

Should have a production release in march or April.

Will close this issue now. Thank you very much!

closed 2021-01-31 by baconpaul

baconpaul, 2021-03-07:
Hello; Just an FYI - the integration with your software is going great. As well as adding Clouds as “Nimbus” we have added Plaits oscillator and called it “Twist” in our nightlies (both to be evocative of the act of twisting hair into Plaits, and also twisting knobs on a physical module). Just wanted to quickly flag that this name was Ok with you also.

Plaits/Twist in Surge sounds awesome. People are pretty excited! (Clouds/Nimbus sounds good too but I still have some work to do to finish the integration). Thank you very much for making this code open source.

pichenettes, 2021-03-07:
Ok, you can proceed with the Plaits integration with the "Twist" name.

baconpaul, 2021-04-23:
Hello. Just a note. We released Surge 1.9 this week with Nimbus (Clouds) as an FX and Twist (Plaits) as an oscillator as well as a collection of other features 

The response has been overwhelmingly positive.

Thank you so much for making this excellent software free and open source.