pichenettes/eurorack #40: No Aux in plaits when harmonics set to 0.5

ypujante, 2021-12-22:
While porting the code to another platform I found an issue with plaits. For engine 0, when harmonics is set to exactly 0.5, Aux remains silent for as long as you play it.

Here is a small program demonstrating the issue:

```cpp
template<size_t N>
std::string toString(plaits::Voice::Frame const iFrames[N])
{
  std::ostringstream str{};
  str << "short out[] = {";
  for(int i = 0; i < N; i++)
  {
    if(i > 0)
      str << ", ";
    str << iFrames[i].out;
  }
  str << "};\n";
  str << "short aux[] = {";
  for(int i = 0; i < N; i++)
  {
    if(i > 0)
      str << ", ";
    str << iFrames[i].aux;
  }
  str << "};\n";
  return str.str();
}

void test()
{
  plaits::Voice voice{};
  char buffer[16384]{};
  stmlib::BufferAllocator allocator(buffer, sizeof(buffer));
  voice.Init(&allocator);

  constexpr int N = 120;
  
  plaits::Voice::Frame frames[120];

  plaits::Patch patch{
    .note = 69,
    .harmonics = 0.5,
    .timbre = 0.5,
    .morph = 0.5,
    .engine = 0,
    .decay = 0.421875,
    .lpg_colour = 0.5
  };

  plaits::Modulations modulations{
    .trigger = 1,
    .trigger_patched = true
  };

  auto ptr = &frames[0];

  for(int i = 0; i < N / 12; i++)
  {
    voice.Render(patch, modulations, ptr, 12);
    ptr += 12;
  }

  std::cout << toString<N>(frames) << std::endl;
}
```

And here is the output:

```
short out[] = {6, 24, 53, 95, 150, 222, 310, 417, 546, 698, 876, 1082, 1161, 1393, 1643, 1902, 2165, 2428, 2690, 2948, 3201, 3449, 3692, 3928, 4227, 4697, 5111, 5470, 5781, 6053, 6292, 6504, 6694, 6865, 7019, 7159, 7363, 7600, 7751, 7853, 7929, 7989, 8037, 8077, 8108, 8132, 8147, 8156, 8125, 8050, 7957, 7861, 7759, 7654, 4889, -1522, -6471, -7927, -8239, -8606, -8829, -9050, -9239, -9439, -9632, -9828, -10023, -10218, -10412, -10606, -10801, -10995, -11187, -11379, -11572, -11766, -11959, -12152, -12345, -12538, -12731, -12924, -13116, -13309, -13499, -13689, -13881, -14073, -14264, -14456, -14647, -14838, -15029, -15220, -15411, -15602, -15790, -15979, -16169, -16359, -16549, -16738, -16928, -17117, -17307, -17496, -17828, -19427, -20127, -8664, 7328, 15713, 17161, 17519, 17380, 17223, 17019, 16823, 16621, 16420};
short aux[] = {1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1};
```

You can change `N` to whatever value you want (`1200` for example) and you will see that Aux always remain to 1..

In my device if you turn the harmonics knob so that it is no longer 0.5 and bring it back to 0.5, then Aux now has sound. You can simulate this with changing the test code like this:

```cpp
  auto ptr = &frames[0];

  patch.harmonics = 0.6;
  voice.Render(patch, modulations, ptr, 12);
  ptr += 12;

  patch.harmonics = 0.5;
  for(int i = 1; i < N / 12; i++)
  {
    voice.Render(patch, modulations, ptr, 12);
    ptr += 12;
  }
```

And you get this output:

```
short out[] = {6, 24, 53, 94, 150, 220, 307, 412, 538, 685, 856, 1052, 1126, 1345, 1580, 1822, 2066, 2309, 2549, 2784, 3013, 3237, 3455, 3666, 3933, 4351, 4717, 5033, 5306, 5544, 5751, 5934, 6097, 6241, 6371, 6487, 6652, 6842, 6958, 7033, 7084, 7122, 7148, 7166, 7176, 7178, 7172, 7159, 7105, 7010, 6904, 6796, 6683, 6566, 3790, -2632, -7592, -9059, -9383, -9761, -9990, -10213, -10402, -10601, -10794, -10989, -11183, -11378, -11572, -11766, -11959, -12153, -12345, -12536, -12729, -12922, -13115, -13308, -13501, -13693, -13885, -14078, -14270, -14461, -14652, -14841, -15032, -15224, -15415, -15606, -15797, -15988, -16178, -16369, -16559, -16749, -16938, -17125, -17315, -17504, -17746, -19158, -20870, -15242, -5128, 914, 2110, 2236, 2576, 7075, 12985, 16020, 16438, 16448, 16270, 16085, 15884, 15684, 15483, 15283};
short aux[] = {1, 1, 0, -1, -4, -8, -14, -20, -28, -36, -45, -53, -55, -72, -90, -110, -131, -155, -180, -207, -234, -263, -292, -323, -944, -1848, -2706, -3638, -4486, -5191, -4938, -4523, -4001, -3387, -2849, -2416, -1896, -1393, -1174, -1087, -1060, -1061, -1076, -1096, -1120, -1144, -1170, -1195, -1223, -1247, -1263, 1683, 11448, 20240, 23089, 23386, 23823, 19446, 8903, 835, -1198, -1262, -1413, -1376, -1399, -1389, -1393, -1390, -1390, -1389, -1389, -1388, -1388, -1387, -1387, -1386, -1386, -1385, -2048, -6501, -11658, -13915, -14171, -14306, -13168, -8406, -3573, -1697, -1514, -1378, -1391, -1375, -1379, -1376, -1376, -1375, -1375, -1374, -1374, -1373, -1373, -1372, -1372, -1371, -1371, -1082, 5718, 16164, 22409, 23976, 24589, 24056, 16323, 6168, 765, -479, -1108, -1295, -1351, -1359};
```

pichenettes, 2021-12-22:
The code produces the expected behavior: the sum of two identical waveforms at the same frequency can be null if their phase difference is 180 degrees.

Note that this does not happen on actual hardware because HARMONICS is never exactly 0.5 forever.

Unless you add noise to this parameter (or more directly to the frequency of the auxiliary_ oscillator), there can always be a position and change of HARMONICS that will cause this dephasing to occur.

closed 2021-12-22 by pichenettes

ypujante, 2021-12-22:
Thank you very much for responding quickly. My device offers a Mix knob that lets you mix Out and Aux, and since the default patch is exactly the one from the test case above, then it is definitely not a great first experience for a user to have no sound when they start tweaking knobs. And of course I cannot explain "this is a feature". I will look into having some "hack" similar to VCV then.. Unless you have another idea for a potential "hack"...

Thank you again!