pichenettes/eurorack #43: plaits / chord engine "pop"

ypujante, 2022-01-03:
When you play the plaits/chord engine with a given set of parameter, there is a distinctive pop at the beginning.

First this is what it looks like (I believe the highlighted section is the cause):

[image]

Second, here is the actual data:

```
    1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,
    1, 1, 1, 1, 1, 1, 1, 1, 1, 50, 99, 150, 199, 231, 269, 303, 355, 406, 413, 444, 494, 525, 569, 609, 656, 702, 747,
    792, 836, 879, 921, 961, 1000, 1025, 1073, 1120, 1168, 1215, 1263, 1312, 1361, 1412, 1464, 1518, 1573, 1623, 1691,
    1759, 1856, 1977, 2056, 2147, 2244, 2321, 2397, 2464, 2520, 2574, 2649, 2711, 2767, 2818, 2867, 2914, 2960, 3006,
    3053, 3100, 3149, 3209, 3285, 3359, 3430, 3499, 3568, 3636, 3706, 3777, 3850, 3925, 4004, 4106, 4267, 4458, 4683,
    4933, 5202, 5489, 5793, 6112, 6446, 6796, 7161, 7646, 8196, 8667, 9110, 9557, 10017, 10325,
   
   // highlighted section
    9350, 7839, 6898, 6657,
    5167, 2146, -2393, -6043, -7766, -8303, -8759, -9146, -10058, -11374, -12437, -13122, -13724, -14921, -18304,
    -21643, -23053, -23440, -23832, -24125, -24438, -24729, -25018, -25297, -25570, -25828, -25911, -25781, -25550,
    -25305, -25070, -24866, -24695, -24582, -24577, -24744, -25062, -25071, -24793, -24501, -24254, -23988, -23752,
    -23562, -23495, -23650, -24038, -24456, -24722, -24916, -25063, -25087, -25043, -25015, -24957, -24860, -24781,
    -24688, -24554, -24431, -24309, -23854, -23285, -22833, -22418, -22003, -21686, -21424, -21290, -21435, -21862,
    -22331, -22631, -22779, -22850, -22831, -22499, -16451, -6008, 978, 2843, 3221, 3432, 3525, 3610, 4098, 4603, 4765,
    4732, 4671, 4510, 4271,

    3880, 3350, 2852, 2447, 2060,};
```

I think that the massive jump `-22499, -16451, -6008, 978` is what most likely causes the pop.. the rest of the waveform is very smooth and does not show this kind of distorted pattern.

Here is the code to reproduce:

```cpp
int main()
{
  char sharedBuffer[16384] = {};
  stmlib::BufferAllocator allocator(sharedBuffer, sizeof(sharedBuffer));
  plaits::Voice voice{};
  voice.Init(&allocator);

  auto patch = plaits::Patch{58, 0.2343, 0.4219, 0.6094, 0, 0, 0, 6, 0.5, 0.5};
  auto modulations = plaits::Modulations{0, 0, 0, 0, 0, 0, 1, 0, false, false, false, true, false};

  plaits::Voice::Frame frames[plaits::kBlockSize];

  for(int i = 0; i < 20; i++)
  {
    voice.Render(patch, modulations, frames, plaits::kBlockSize);

    for(int j = 0; j < plaits::kBlockSize; j++)
    {
      std::cout << frames[j].out << ", ";
    }
  }
  std::cout << std::endl;
}
```

pichenettes, 2022-01-03:
In the code example you have posted, the pop is not caused by a massive jump in negative values (which are valid numbers for a 16-bit signed PCM sample), but simply because the Chord model has a low-pass filter on both the .morph and .timbre parameters - initialized with a value of 0.0f. https://github.com/pichenettes/eurorack/blob/master/plaits/dsp/engine/chord_engine.cc#L208

So effectively, they are quickly swept from 0.0 to 0.4 and 0.6 in the first 10 blocks or so.

If you intend to use this code for the offline rendering of one-shot samples and expect to get valid results right from the first block, you can either remove the low-pass filters on the parameters or add logic to directly initialize the state of the filters to the value of the parameters for the first invocation of Render() – or alternatively, you can discard the first dozen blocks during which all variables carrying the state of the generator will be stabilizing.

These are not concerns and issues with the Eurorack module Plaits - which is not expected to generate accurate audio in the first milliseconds after it is powered on. There are much more noticeable pops anyway when the DAC and output op-amp receive power.

closed 2022-01-03 by pichenettes

pichenettes, 2022-01-04:
You're going to run into trouble for all the models making use of random numbers – this is probably what's happening with the Swarm engine.

I don't have any solution to your problem - the module has certainly not be designed with the goal of making each "note" sound exactly the same.

ypujante, 2022-01-05:
@pichenettes sorry to bother you again. After digging more into the code, there is something that still seems fishy to me.

[image]

If you look at this picture it shows:

The top waveform is what comes out of the voice
The bottom waveform is what comes out of the engine (the swarm engine in this case)

There does not appear to be anything wrong with the swarm engine itself but it seems to me that the processing in the voice itself is distorting the waveform in a non natural way in the first few ~20 batches or so (about a quarter of the picture). I believe now it was the same issue I was seeing with the chord engine.

I know it does not affect the hardware, but is there anything that you can think of that would trigger this distortion?

pichenettes, 2022-01-05:
The code is behaving as expected in this example.

Because of the beating between the different oscillators that make up the SwarmEngine, and also because the interplay of the TIMBRE and MORPH knobs can create textures with sparse grains (and gaps of silence between them), there can be large variations in amplitude in the resulting waveform. To make it sound more solid and reduce these variations, the signal is amplified by a gain of 3 and processed by a limiter (effectively applying the gain of 3 to only the quieter sections). This limiter employs a low pass filter to detect the peaks of the signal and has a warm-up period after it has been initialized. You can disable this gain and the limiter by changing this line:

https://github.com/pichenettes/eurorack/blob/master/plaits/dsp/voice.cc#L47

to:

engines_.RegisterInstance(&swarm_engine_, false, 1.0f, 1.0f);

ypujante, 2022-01-06:
I tried your suggestion but it doesn't really help with the problem. So I guess that wasn't the issue. I think the true issue is that the waveform "repeats" and it seems that the very first repetition starts in the middle as you can see in this screenshot: the second part of the waveform shows 7 peaks. The first part shows only 4 which seems to indicate that it starts in the middle.. hence the pop.

[image]

pichenettes, 2022-01-06:
Your question was why there is a difference in waveform between the raw output of Engine and the output of Voice, and I provided you with an explanation: the action of the limiter applied in the post-processing stage of a voice.

I'm afraid I won't have more time to help you with your personal programming project.

ypujante, 2022-01-06:
Sorry to have bothered you