pichenettes/eurorack #45: Add Beads

electron271, 2022-03-05:
Add Beads source code and hardware design

pichenettes, 2022-03-05:
Not yet.

closed 2022-03-05 by pichenettes

Harry-Harrison, 2023-01-23:
@pichenettes Any plans for this one again?

Harry-Harrison, 2023-03-01:
Not sure why the down-thumbs for my question, it's no longer possible to buy a Beads module unless you're willing to pay a jacked up price due to the rarity.

modularev, 2023-03-01:
it's still available from retailers, you just have to look around a bit.

Harry-Harrison, 2023-03-01:
@modularev What country? I'm in the UK and none of the retailers have it and I can't find any in europe with it either (though even that would cost an extra chunk of money on import fees.

KittenVillage, 2023-03-01:
Well just because you haven't bought one by now is no reason to expect the creator to release their design. You do realize that they are under no obligation to release it, and it's rather rude to ask.

Harry-Harrison, 2023-03-01:
Hey gang, I totally agree with you! I just took "Not yet" to mean "One day" and figured the announcement of ceasing production on the official models would be a good time to ask. I'm only recently getting into modular (just weeks before that announcement).

In case it wasn't obvious, no offence was meant to anyone by my asking if plans had changed.

ramonauble, 2023-03-03:
@Harry-Harrison in case you were trying to get one at a reasonable price (as opposed to the ridiculous and arbitrarily inflated prices of shameless module scalpers), robotspeak has 2 in stock at list, and there are a handful on reverb in the $300-$400 range. granted, most wouldn't be local (to you), but with shipping and import tax, probably would still be considerably less than what some people are asking for them.

berlingaytwink, 2023-09-05:
I'm wondering if Beads ever gets added to the repository.
I feel super sorry that Behringer copied your modules for super cheap :( if thats the reason maybe add a license to exclude them from releasing it.
Im mainly interested in the code for the delay and reverb which i love <3 
You did so much for the audio programmer community. So thanks!