pichenettes/eurorack #47: Convert EAGLE files to XML

electron271, 2022-06-19:
I know that this won't benifit MI much, but considering that MI is shutting down (bye pichenettes, I'll really miss you, you have been an amazing figure in the Eurorack community), I think converting them to XML would make it easier for people to study the MI products (this way you can import into something like KiCAD).

electron271, 2022-06-19:
I have EAGLE free on my other computer, so I can try converting some of them myself and then open a PR

electron271, 2022-06-27:
@pichenettes, I'll install EAGLE on my computer and make a PR for this issue, I'll send the number when I start it

electron271, 2022-06-27:
#48

electron271, 2022-07-28:
Will be closed by #48

luzpaz, 2023-07-31:
KiCAD files are also an option.

nissaba, 2024-08-09:
Eagle is dead (no longer free, and only available as part of Fusion 360), the PCB files are in a binary format and it would be useful for everyone to update the PCB file to the XML format so that software like Kicad can read it.

forestcaver, 2024-08-09:
> Eagle is dead (no longer free, and only available as part of Fusion 360), the PCB files are in a binary format and it would be useful for everyone to update the PCB file to the XML format so that software like Kicad can read it.

If you do that, how do you verify the files ? You are asking pichenettes to essentially build every module with the new files to verify them which is unreasonable or to verify them in some other way (can think of some but it’s a *lot* of work). If you want to use kicad then why not just import the eagle files yourself or just use the files from the pull request !???

nissaba, 2024-08-09:
> If you do that, how do you verify the files ? You are asking pichenettes to essentially build every module with the new files to verify them which is unreasonable or to verify them in some other way [...]

I’ve encountered an issue with KiCad where it refuses to accept files in binary format, as it requires them to be in XML format. I tried this myself earlier today, and unfortunately, KiCad was unable to read the files.

Given that the project has 722 forks, I was hoping that pichenettes (the original creator) had received some pull requests to address this issue. However, after speaking with my co-worker, who is an electronics/electrical engineer, I learned that while Eagle is essentially discontinued, it is now integrated into Fusion 360 as part of a paid subscription for the full-featured version.

Interestingly, Fusion 360 for personal use does include a working version of Eagle. To access it, you need to go to the File menu and select the new “Electronic Design” option. From there, you can import files from the PCB folder.

My hope is that we can use Fusion 360 to convert the files to a non-binary (XML) format and test whether KiCad can then accept them.

So, Eagle isn’t really dead—it has just been absorbed into Fusion 360.

btomwhitehead, 2024-08-09:
> > > Eagle is dead (no longer free, and only available as part of Fusion 360), the PCB files are in a binary format and it would be useful for everyone to update the PCB file to the XML format so that software like Kicad can read it.
> > 
> > 
> > If you do that, how do you verify the files ? You are asking pichenettes to essentially build every module with the new files to verify them which is unreasonable or to verify them in some other way (can think of some but it’s a _lot_ of work). If you want to use kicad then why not just import the eagle files yourself or just use the files from the pull request !???
> 
> I’ve encountered an issue with KiCad where it refuses to accept files in binary format, as it requires them to be in XML format. I tried this myself earlier today, and unfortunately, KiCad was unable to read the files.
> 
> Given that the project has 722 forks, I was hoping that pichenettes (the original creator) had received some pull requests to address this issue.
> 
> However, after speaking with my co-worker, who is an electronics/electrical engineer, I learned that while Eagle is essentially discontinued, it is now integrated into Fusion 360 as part of a paid subscription for the full-featured version.
> 
> Interestingly, Fusion 360 for personal use does include a basic version of Eagle. To access it, you need to go to the File menu and select the new “Electronic Design” option. From there, you can import files from the PCB folder.
> 
> My hope is that we can use Fusion 360 to convert the files to a non-binary (XML) format and test whether KiCad can then accept them.
> 
> So, Eagle isn’t really dead—it has just been absorbed into Fusion 360.

When you open EAGLE (standalone) theres a nice big heading saying:

> On June 7, 2026, EAGLE will no longer be available - [click here for more information](https://www.autodesk.com/products/fusion-360/blog/future-of-autodesk-eagle-fusion-360-electronics/?mktvar002=5825936031&utm_medium=product&utm_source=ipm&utm_campaign=5825936dmfusion-360-and-eagle---unification-fy24&utm_id=5825936031)

I would recommend reading such notices.

nissaba, 2024-08-09:
> When you open EAGLE (standalone) theres a nice big heading saying:
>
>
> I would recommend reading such notices.

In my case Eagle is gone from the system as I had to reformat my HD (so I can not get the warning you are talking about ), afther witch I was only able to install Fusion to do some 3d modelling. I was not aware of the changes as I had not opened Eagle in the last 2 years. When I tried to get re-install Eagle back I was hitting the web page informing that is will no longuer be availeble out side of the Fisuin 360 subscription. And Since I had always used Eagle as a stand alone software it was not clear that it would be inbedded in the Fission app.

dessertplanet, 2026-09-16:
For the next person looking for this: You might try cloning the upstream repo, then running `gh pr checkout 47` inside the repo