pichenettes/eurorack #50: where is the file---"bootloader.hex"?

LegendaryTeacherLiu, 2022-10-21:
i am first to diy the mutable eurorack rings. i use st-link utility to upload the firmware. but it is only" *.hex f"ile can be use. i can't find "bootloader.hex". just only find "bootloader.cc". how can i do ? how to make *.hex?

gorcha, 2022-10-21:
Hi @LegendaryTeacherLiu, you need to compile the bootloader using the included makefile to produce the .hex file.

Check out the [vagrant dev environment](https://github.com/pichenettes/mutable-dev-environment) for a fully configured build environment. There are also instructions for building on OSX with homebrew in issue #28.

LegendaryTeacherLiu, 2022-10-21:
Thank you for your reply. I will try to do it. and Can I directly use the wav file on the mutable website to complete the firmware upgrade?

gorcha, 2022-10-21:
Yep, you can update the firmware using the files available on the website [here](https://mutable-instruments.net/modules/rings/firmware/).

The manual has instructions for performing the update from a wav file:
https://mutable-instruments.net/modules/rings/manual/#firmware

LegendaryTeacherLiu, 2022-10-21:
Thank you again. Because I don't have the ability to program. then I will use the wav file on the website to upgrade the firmware.

pichenettes, 2022-10-21:
This online is space is for reporting issues in the codebase, this is certainly not a support forum!

closed 2022-10-21 by pichenettes