pichenettes/eurorack #51: Missing file "variable_shape_oscillator.h" when compiling the target stages

Leaumonte, 2023-01-04:
Trying to build the target stages with this command

`make -f stages/makefile`

Leads to this following error

In file included from ./stages/chain_state.h:35:0,
                 from stages/chain_state.cc:29:
./stages/segment_generator.h:39:46: fatal error: stages/variable_shape_oscillator.h: No such file or directory
 #include "stages/variable_shape_oscillator.h"
                                              ^
compilation terminated.
make: *** [build/stages/chain_state.o] Error 1

pichenettes, 2023-01-04:
> Did you pull the latest updates and submodules?
>
> git pull git submodule init git submodule update
>
> There are references from chain_state.h to segment_generator.h, which again uses files from tides2 and from stmlib.

Can I get a couple of days of rest???????

Sent on the go

Leaumonte, 2023-01-04:
> Did you pull the latest updates and submodules?
> 
> `git pull` `git submodule init` `git submodule update`
> 
> There are references from chain_state.h to segment_generator.h, which again uses files from tides2 and from stmlib.

Yes all these git commands are running by the [mutable-dev-environment ](https://github.com/pichenettes/mutable-dev-environment/blob/master/1_clone-code.sh)

Leaumonte, 2023-01-09:
Fixed by this commit
https://github.com/pichenettes/eurorack/commit/897806facd8fd9f696585a4b83ec81d9e900f5a2

closed 2023-01-09 by Leaumonte